victronenergy/gui-v2 #2248: Venus OS 3.60: GUI-V2 fails to load

ReinvdZee, 2025-06-17:
Reported by a customer with an Ekrano. see https://community.victronenergy.com/t/firmware-v3-60-ui-not-loading/36087

After updating to 3.60, the UI fails to load. I checked the logs:

```
2025-06-17 02:43:30.415287500 *** CCGX booted (0) ***
2025-06-17 02:43:30.933533500 *** starting start-gui ***
2025-06-17 02:43:31.382077500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-06-17 02:43:32.469183500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-06-17 02:43:33.571951500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-06-17 02:43:34.646243500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-06-17 02:43:35.716078500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-06-17 02:43:37.576061500 method return time=1750128217.575785 sender=:1.29 -> destination=:1.35 serial=12 reply_serial=2
2025-06-17 02:43:37.576071500    int32 0
2025-06-17 02:43:38.516856500 Victron gui version: v1.1.20
2025-06-17 02:43:39.548229500 Connecting to system bus...
2025-06-17 02:43:39.597017500 Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
2025-06-17 02:43:39.862571500 BackendConnection state: Victron::VenusOS::BackendConnection::Ready
2025-06-17 02:43:39.908435500 venus.gui: Successfully loaded translations for locale "en_US"
2025-06-17 02:43:39.911900500 venus.gui: Unable to load translations for locale "C"
2025-06-17 02:43:39.923608500 venus.gui: Falling back to English as locale catalogue failed to load.
2025-06-17 02:43:51.806611500 qml: Cannot load temperature unit, dbus/com.victronenergy.settings/Settings/System/Units/Temperature has unsupported value:  default to celsius
2025-06-17 02:43:51.823163500 qml: Cannot load altitude unit, dbus/com.victronenergy.settings/Settings/System/Units/Altitude has unsupported value:  default to meter
2025-06-17 02:43:55.744826500 qml: Loading D-Bus data backend...
```

That's all the logging there is while it was running 3.60. After a few hours, the user downgraded to 3.55 and everything worked normal again.

chriadam, 2025-06-17:
We don't load the UI until all of the DataManager.qml's `_dataObjectsReady` property goes to true.  In this case, I believe it is failing because the `Global.systemSettings` object is never completing initialisation, as based on the log it looks like the system is missing the systemsettings service.

chriadam, 2025-06-17:
I didn't know that such a system was possible.  We use systemsettings for a lot of important things.  What should we do here?  Just assume some default values for every possible system setting?  Or?

ReinvdZee, 2025-06-17:
@chriadam Can it be that localsettings is just not up when the GUI starts? And does the GUI report that the connection has been established at last? Because logs also show these messages when the unit boots 3.55 and the UI loads fine:

```
2025-06-17 06:14:09.989616500 *** CCGX booted (0) ***
2025-06-17 06:14:10.319664500 *** starting start-gui ***
2025-06-17 06:14:10.506857500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-06-17 06:14:11.572709500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-06-17 06:14:12.671276500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-06-17 06:14:13.728037500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-06-17 06:14:14.798205500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-06-17 06:14:16.007942500 method return time=1750133656.007037 sender=:1.30 -> destination=:1.34 serial=15 reply_serial=2
2025-06-17 06:14:16.007959500    int32 0
2025-06-17 06:14:16.630987500 Victron gui version: v1.0.9
2025-06-17 06:14:16.638123500 Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
2025-06-17 06:14:16.638140500 Qt depends on a UTF-8 locale, but has failed to switch to one.
2025-06-17 06:14:16.638142500 If this causes problems, reconfigure your locale. See the locale(1) manual
2025-06-17 06:14:16.638146500 for more information.
2025-06-17 06:14:17.405047500 Connecting to system bus...
2025-06-17 06:14:17.413695500 Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
2025-06-17 06:14:17.479701500 BackendConnection state: Victron::VenusOS::BackendConnection::Ready
2025-06-17 06:14:17.487906500 venus.gui: Successfully loaded translations for locale "en_US"
2025-06-17 06:14:17.488824500 venus.gui: Unable to load translations for locale "C"
2025-06-17 06:14:17.489080500 venus.gui: Falling back to English as locale catalogue failed to load.
2025-06-17 06:14:28.439906500 qml: Cannot load temperature unit, dbus/com.victronenergy.settings/Settings/System/Units/Temperature has unsupported value:  default to celsius
2025-06-17 06:14:29.263939500 UID "0/NumberOfPhases" contains invalid part: "0"
2025-06-17 06:14:29.419767500 qml: Loading D-Bus data backend...
2025-06-17 06:14:35.106450500 file:///opt/victronenergy/gui-v2/Victron/Dbus/data/dbus/SolarChargersImpl.qml:20:13: QML SolarCharger: Deprecated behavior! Device serviceUid already set to dbus/com.victronenergy.solarcharger.ttyUSB13, should not be changed again todbus/com.victronenergy.solarcharger.ttyUSB13
2025-06-17 06:14:36.219447500 file:///opt/victronenergy/gui-v2/Victron/Dbus/data/dbus/SolarChargersImpl.qml:20:13: QML SolarCharger: Deprecated behavior! Device serviceUid already set to dbus/com.victronenergy.solarcharger.ttyUSB14, should not be changed again todbus/com.victronenergy.solarcharger.ttyUSB14
2025-06-17 06:14:36.239292500 file:///opt/victronenergy/gui-v2/Victron/Dbus/data/dbus/EnvironmentInputsImpl.qml:19:13: QML EnvironmentInput: Deprecated behavior! Device serviceUid already set to dbus/com.victronenergy.temperature.ruuvi_da4d4d8804cb, should not be changed again todbus/com.victronenergy.temperature.ruuvi_da4d4d8804cb
2025-06-17 06:14:38.168691500 qt.virtualkeyboard: Cannot create directory for user data "//.config/qtvirtualkeyboard"

......

```

chriadam, 2025-06-18:
You're right, it's not systemsettings.

From the log, we see `2025-06-17 02:43:55.744826500 qml: Loading D-Bus data backend...` and that line would only be printed if all of the data objects are ready.  So, it is not what I originally thought.

It sounds like the problem is something to do with the transition to hide the Splash View (it must be getting stuck somewhere, and not completing, perhaps?)

I've created https://github.com/victronenergy/gui-v2/pull/2256 which adds some debugging, but honestly I'm not sure why it would fail for this guy but not for anyone else.  The only thought I had is that the screen blanker "read" call might be blocking infinitely?  So the UI never finishes loading, and so the splash screen never fades itself?  But not sure how to test this, without running a custom build on his device :-/

ReinvdZee, 2025-06-18:
@chriadam Thanks for looking into this. I've created an L3 ticket and asked if its ok to install 3.60 again and make some modifications to the GUI.

mpvader, 2025-06-19:
@chriadam & @ReinvdZee , Rein wrote:

> Can it be that localsettings is just not up when the GUI starts?

I’m 99.9% sure that gui-v2 is only started *after* localsettings is up.

Login to a GX and check the startup scripts to be sure.

ReinvdZee, 2025-06-19:
@mpvader You're right, the `start-gui` script waits until it is able to read the value of `/Settings/Gui/RunningVersion` before starting the GUI.

chriadam, 2025-06-20:
I have created two PRs (one for main branch, one for v3.60 branch) which fix a couple of issues, do some defensive error handling in some potential code paths (e.g. blocking reads in the screenblanker), and add a whole heap of debugging so that we can identify the point of failure in future if there is still some problem.

https://github.com/victronenergy/gui-v2/pull/2256
https://github.com/victronenergy/gui-v2/pull/2273

I've tagged v3.60 branch with tag v1.1.22 which contains the changes.

mpvader, 2025-06-20:
Released as public beta v3.62~1

chriadam, 2025-07-31:
Customers are still reporting an issue, even with Qt 6.8.3 builds.

chriadam, 2025-11-06:
Pushing this back to v1.3.0 as we don't have a clear plan of attack for resolving this one, currently.

chriadam, 2026-04-14:
No new reports since June 2025, and v3.70 and v3.71 have been released since, I believe this is fixed.  If not, we can open a new task.

closed 2026-04-14 by chriadam