victronenergy/gui-v2 #2251: Venus OS | CerboGX | gui-v2 : Fronius Yield Today display shows not interesting valus "Total Energy"

uelpenich, 2025-06-17:
in cerbogx gui-v2 Solar **yield** | **Overview** | **Solaryield** display **column Yield Today** the line PV-Inverter | Fronius Garagendach (Fronius Symo 6.0-3-M) shows the column "**Energy**". The value is the **Energy total** produced by the Fronius Inverter since the birthday of the inverter (maybe since the last factory reset). 
I would prefer to see as with the victron MPPTs the **Yield Today** column and value. 

Nice to have:
a sum line for the columns **Yield today** and **Power**. 

[image]

izak, 2025-06-18:
Indeed, I've mentioned before that this figure isn't very useful. If the PV inverter has been in operation a really long time, it becomes just a large meaningless number.

Unfortunately the Sunspec standard, which is what we use to talk to PV inverters, does not have provision for "yield today" as we have with Victron products. I would vote for just removing it.

uelpenich, 2025-06-18:
My idea is to sumarize in CerboGX / Venus the Fronius **Power** information to achieve the **Yield Today** value. 

Question: 
Where are the **Yield Today** values sumarized, in the MPPT devices or in CerboGX / Venus?

mpvader, 2025-06-19:
ok lets remove it.

@uelpenich : I think what you're proposing is that the GX itself maintains / calculates what the yield today is; which would be a nice solution but something we're planning to do anytime soon.

blammit, 2025-08-28:
I'm reworking the page model as part of #2371 so I just remove this data point at the same time.