victronenergy/gui-v2 #2277: Temperature sensor value from BMS has generic name that does not identify it as belonging to battery

blammit, 2025-06-22:
See https://community.victronenergy.com/t/venus-os-v3-62-1-available-for-public-beta-testing/36621/4

Would be better if this said “Pylontech US5000 BMS Temperature” or similar.

ReinvdZee, 2025-06-24:
This feature was designed for use with Victron NG batteries, and for those it works fine:

[image]

As expected. the customname is used to display the temperature service.

blammit, 2025-07-01:
I can reproduce this when a "custom name" is not set for a battery/temperature device. The problem is that in this case, gui-v2 does not fallback onto the product name.

blammit, 2025-07-03:
@ybott-qinetic This is reproducible on https://vrm.victronenergy.com/installation/133622/dashboard:

[image]

The "Battery temperature sensor (22)" should show the product name of the service instead.

commit 3ee494d9 (2025-07-09): Temp Sensor: Use product name if custom name is unavailable

Use device name to ensure the product name is shown if the custom
name is unavailable for the temperature sensor.

Fixes #2277
PR #2318 (2025-07-09): Temp Sensor: Use product name if custom name is unavailable
Use device name to ensure the product name is shown if the custom name is unavailable for the temperature sensor.

Fixes #2277
commit d67e4bea (2025-07-09): Temp Sensor: Use product name if custom name is unavailable

Use device name to ensure the product name is shown if the custom
name is unavailable for the temperature sensor.

Fixes #2277
closed 2025-07-09 by ybott-qinetic