victronenergy/gui-v2 #2281: DC load / fridge labeled as DC generator

ThiemoVanEngelen, 2025-06-23:
When a Smartshunt/BMV is configured as DC load of type fridge, it is labeled in the device page (via the overview load box or the device list) as DC generator

[image]

A system available for testing is: https://vrm.victronenergy.com/installation/50884/dashboard

Reported via the community: https://community.victronenergy.com/t/smart-shunt-type-fridge-appears-as-dc-generator-in-remote-console-v3-60/36782

mpvader, 2025-06-28:
Hi, this is a bug - right? even though small, still a bug and probably simple to fix - so listed it as high prio.

ReinvdZee, 2025-07-01:
@blammit The list can be found in [velib](https://github.com/victronenergy/velib/blob/master/data/protocols/vregs.xml#L2927-L3001)

But I've updated the dbus API wiki as well: https://github.com/victronenergy/venus/wiki/dbus#dcload

This reminds me of an [investigation](https://github.com/victronenergy/venus/issues/1223#issuecomment-2437180897) I did earlier about the different energy metering modes for the smartshunt. Check the "gui-v2" column, the smartshunt is shown as "DC generator" for nearly all dc load energy metering modes.

blammit, 2025-07-01:
@ReinvdZee oh nice, that is super useful. I will fix gui-v2 to align the naming for all the of the different DC meter types. For the cases that conflict ("wind charger" vs "wind generator" vs "DC source"), what do you suggest? Should I go for the VC terminology?

blammit, 2025-07-04:
> For the cases that conflict ("wind charger" vs "wind generator" vs "DC source"), what do you suggest? Should I go for the VC terminology?

As discussed in the meeting, update gui-v2 to use the VC terminology in all cases.

blammit, 2025-07-15:
The Overview drilldown is also showing the incorrect page in this case; I've filed https://github.com/victronenergy/gui-v2/issues/2338 to fix this.

blammit, 2025-07-21:
@ReinvdZee @philipptrenz as a result of these fixes, `dcgenset` services are now called "Generic source" in the UI. Should these ones still be called "DC generators" instead?

commit e2f59ce7 (2025-07-14): Add dcload services based on Monitor Type

Change some DC Generator references to generic terminology
Fixes #2281
commit bd033607 (2025-07-15): Add dcload services based on Monitor Type

Change some DC Generator references to generic terminology

Fixes #2281
commit 4908ba9a (2025-07-15): Update DC meter names and add names for dcload services

Align DC meter names with those for VictronConnect, and add names for dcload services based on MonitorType.

Fixes #2281
commit c25aedbd (2025-07-15): Update DC meter names and add names for dcload services

Align DC meter names with those for VictronConnect, and add names
for dcload services based on MonitorType.

Fixes #2281