victronenergy/gui-v2 #2302: Fix dbus type conversion issue

mpvader, 2025-07-02:
details here: https://github.com/victronenergy/venus-private/issues/558

https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1751310373519149

chriadam, 2025-07-07:
I've created https://github.com/victronenergy/veutil/pull/17 which cherry picks a fix from Jeroen into the `gui-v2` branch in the veutil submodule (plus a couple of other checks to prevent warning spam when running with the mock backend).

Once that one is merged into the `gui-v2` branch of veutil, I will create the PR in gui-v2 main.

Then, I will cherry pick that same veutil commit into the `gui-v2-v3.60` branch of veutil, and create a commit in `v3.60` branch of gui-v2 repository to update the submodule to point to that one.

chriadam, 2025-07-07:
Done.  Closing as fixed.