victronenergy/gui-v2 #2303: Switch pane design improvements and additions

ReinvdZee, 2025-07-02:
There are some updated and new designs to be added to the switching pane:

Updated basic elements:

[image]
(Note that "Latching" is recently renamed to "Toggle")

New designs:

[image]

The design appear in the order of priority.

Tasks:

- [ ] Implement updates for the dimming and momentary switch.
- [ ] (Rein) Update dbus API wiki.
- [ ] (Dirk-Jan) Update NR virtual switch to support the new types.
- [ ] Implement the new designs for dimmable outputs.

blammit, 2025-07-21:
Not yet completed, reopening.

blammit, 2025-08-22:
@ReinvdZee Please note, instead of using these `Type ` values:

5 = Dropdown
6 = Basic slider
7 = unranged setpoint
8 = 3 state switch

we are currently using these values instead:

6 = Dropdown
7 = Basic slider
8 = unranged setpoint
9 = 3 state switch

This was because "5" was previously reserved for the ES SmartSwitch "slave mode" feature, in case that is still required.

blammit, 2025-09-15:
> It seems more user friendly to only show it once and also include the Customname (if set) in the warning.

Is that necessary, if the warning is triggered by a user clicking on the control anyway? Maybe just showing the warning once will be enough to clear this up.

ReinvdZee, 2025-09-15:
@blammit, @dirkjanfaber,

Regarding the toast message, we should definitely not show more than one message, and can we have the toast automatically dismiss after a few seconds?

wrt the customname in the toast; I agree that it may be more clear, but can also lead to difficulties when the customname is long.
Is there currently a limit for the customname, and would that still fit if the toast was reworked into something like: 

"Unable to control <customname>, set to Auto mode"

chriadam, 2025-11-13:
What remains for this one?  Can we discuss in the meeting?

commit 07e8b2b8 (2025-07-17): Distinct on/off selected backgrounds for Toggle Switches

Partially fixes #2303
commit 31a1112b (2025-07-17): Distinct on/off selected backgrounds for Segmented Buttons

Modified label font size

Partially fixes #2303
commit 491f0925 (2025-07-17): Distinct on/off selected backgrounds for Segmented Buttons

Also modified label font size

Partially fixes #2303
commit bd33c0de (2025-07-28): Distinct on/off selected backgrounds for Segmented Buttons

Also modified label font size

Partially fixes #2303
closed 2025-11-13 by chriadam