victronenergy/gui-v2 #2311: Drill-down for "AC Loads" and "Essential Loads" on Overview page

philipptrenz, 2025-07-03:
See https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1751545755283939

philipptrenz, 2025-07-03:
Also related to https://github.com/victronenergy/venus-private/issues/380

ReinvdZee, 2025-07-10:
As discussed in the meeting, the voltage and frequency per phase can be fetched from the inverter/charger service. 
Systemcalc provides these paths (this is on a system with a multi):

```
Ac/In/0/ServiceName   com.victronenergy.vebus.ttyS4
Ac/In/0/ServiceType    vebus
```

Which tells you where to look for the voltage and frequency values.

philipptrenz, 2025-07-18:
For when to enable the drill-down (clickable blue framed tile on the Overview), I have the following suggestions:

- For "AC Loads" and "Essential Loads" on 1-phase systems, only enable the drill-down if there is at least one service to show
- For "AC Loads" and "Essential Loads" on 3-phase systems, always enable them. The reason is, that the individual phase info might still be insightful
- For "DC Loads", enable them if there is either at least one `.dcload` or `.dcdc` system to show or more than one `.dcsystem` service, so that the table in the top "Total" is insightful again.

blammit, 2025-07-22:
> Which services to list for AC Loads and Essential Loads?
> com.victronenergy.acload
> com.victronenergy.heatpump
> com.victronenergy.evcharger

Currently, if there are both AC Loads and Essential Loads:
- AC Loads shows heat pumps with `/Position=1` (i.e. AC input position)
- Essential Loads shows heat pumps with `/Position=0` (i.e. AC output position)

If we add `acload` and `evcharger` to the services to be shown, then:
- I guess `acload` services are always shown in AC Loads instead of Essential Loads? Is that right?
- `evcharger` services can have a `/Position` value, so we can use that to determine whether to show each charger in AC Loads vs Essential Loads. (Perhaps fall back to AC Loads if `/Position` is invalid/unavailable)

philipptrenz, 2025-07-22:
> Currently, if there are both AC Loads and Essential Loads:
> 
> AC Loads shows heat pumps with /Position=1 (i.e. AC input position)
> Essential Loads shows heat pumps with /Position=0 (i.e. AC output position)
> If we add acload and evcharger to the services to be shown, then:
> 
> I guess acload services are always shown in AC Loads instead of Essential Loads? Is that right?
> evcharger services can have a /Position value, so we can use that to determine whether to show each charger in AC Loads vs Essential Loads. (And fall back to AC Loads if /Position is invalid/unavailable?)

@blammit If there are both AC Loads and Essential Loads, for all three services the correct drill-down needs to be determined by the `/Position` path.

While looking into this, I discovered some alignment issues and will have to wait for some feedback what the correct meanings are for the `/Position` path of `com.victronenergy.acload`, `com.victronenergy.heatpump` and `com.victronenergy.evcharger`. I will come back to you about this as soon as possible.

If we have both AC Loads and Essential Loads and a service is missing the `/Position` path or its value is invalid, using AC Loads as a fallback is fine.

Please also note, that the `/Position` path for the `.acload` and `.heatpump` service is a relatively recent feature, older D-Bus recordings or mocks might not yet have it.

> Also, does clicking on a device in the list take you to its settings page? If so, normally there would be an arrow for it, like this for the solar drilldown when there are multiple devices:
> [...]
> I can add that, it just means the top "total power" and the phase table will get some extra padding on the right, so that they still line up with the power columns in the list of devices.

This is was not considered in the design, probably because of the planned buttons per entry – which would be UX-wise conflicting with a touch gesture for selecting an entry. But as I'm not yet sure if we want these buttons at this position eventually, I also really like the idea of having the devices selectable with the chevron as indicator on the right. My only concern is, that it will be a bit tight on the 5" screen. Maybe you can evaluate that and send me a screenshot?

blammit, 2025-08-05:
PR to add drilldowns for AC Loads and Essential Loads: https://github.com/victronenergy/gui-v2/pull/2402

DC Load drilldowns will be done in a separate PR.

philipptrenz, 2025-09-07:
@blammit I noticed, that on a single-phase system the table is shown, which is not desired. The table is only to be shown on 3-phase systems. 

[image]

blammit, 2025-09-07:
@philipptrenz Thanks for that, have fixed it here: https://github.com/victronenergy/gui-v2/pull/2480

Also included a fix to not show the Essential Loads drilldown at all in the case where it is single-phase and there are no devices to be shown.

philipptrenz, 2025-10-16:
Hi @blammit  & @ybott-qinetic, what's the status of the DC Loads drilldown? Is there already something ready for being reviewed?

ybott-qinetic, 2025-10-21:
@philipptrenz There is a PR https://github.com/victronenergy/gui-v2/pull/2509 I need to complete.

blammit, 2025-11-07:
@philipptrenz A question regarding the "status" text from the `SwitchableOutput` API, as per this earlier comment:

> com.victronenergy.acload: If it has a SwitchableOutput API exposed, you can get a status from there. Otherwise, there is only the total power reading → see [here](https://github.com/victronenergy/venus/wiki/dbus#grid-and-acload-and-genset-meter)
com.victronenergy.heatpump: Same as com.victronenergy.acload (for now)
> com.victronenergy.dcload: If it has a SwitchableOutput API exposed, you can get a status from there. If /Dc/0/Temperature is valid, show this temperature → see [here](https://github.com/victronenergy/venus/wiki/dbus#dcload)

The "status" is provided per-output, like `/SwitchableOutput/<x>/Status`, so there is no overall status for the service, and I assume we can't just show the status from the first available output. Is there something else we should show instead? Or maybe just omit it for these services?