victronenergy/gui-v2 #2334: VE.Bus: Active AC input shows state "AC input 241"

vejpasop, 2025-07-14:
Customer reported "AC input 241" state on the VE.Bus device page of GUI_V2 when AC input was disconnected. 
CCGX GUI (V1) properly reported "Disconnected".

I was able to reproduce this only once on a CCGX wit Venus` V3.63~1 `by disconnecting AC in.
`V3.55` and `V3.70~9` seemed to work ok. When reverting back to V3.63~1 I could no longer get it to show "AC input 241" 
The VRM data during the 241 occurrence  `com.victronenergy.vebus.ttyO1/Ac/ActiveIn/ActiveInput` logged a "disconnected" state.

The data on `Ac/ActiveIn/ActiveInput` is defined as Active input: 0 = ACin-1, 1 = ACin-2,  240 is none.

So probably the GUI added a +1 offset to the 240 state.

Hints:
Is there a dependency on other dbus items in order to show "Disconnected"?

The first and only occurrence was within a few minutes after booting V3.63~1
At that time it was accompanied by a "grid lost" notification.

[image]

blammit, 2025-07-15:
This is a duplicate of https://github.com/victronenergy/gui-v2/issues/2269, which is fixed in the latest beta. The problem was the code did not check for the "240" state (i.e. when inverting). So it now shows "Disconnected" as expected in that case.

closed 2025-07-15 by blammit

mpvader, 2025-07-15:
https://community.victronenergy.com/t/victron-mp2-anzeige-mit-cerbo-v3-62-anzeige-merkwurdig/38297/1