victronenergy/gui-v2 #2374: Side pane can't be closed

mpvader, 2025-07-23:
No matter how often I click on the icon on the top right, side pane doesn't close.

Its a system without any multi or other common victron device, has only some GX IO Extenders connected and probably node-red flows, perhaps that has something to do with it.

[image]

[image]

blammit, 2025-07-24:
The icon in the top-right indicates the UI thinks the side pane is currently closed, so that's why it cannot be closed in this state. Needs further investigation.

blammit, 2025-07-24:
We were able to reproduce this when:

- "Brief page (Side panel open)" was set as the start page
- Animations were disabled (this may or may not be relevant)

commit 1adb87cc (2025-08-25): Brief: allow side panel to be closed when Brief is the Start Page

The side panel can only be closed when state="panelOpened", and this
state is set by the ScriptAction in the "initialized" -> "panelOpening"
transition. So this transition must be run whenever the state becomes
"panelOpening", regardless of whether Global.allPagesLoaded is true;
otherwise, it is possible to get stuck in the "panelOpening" state
without moving onto the "panelOpened" state.

This was happening when Brief was loaded as the Start Page, and
Global.allPagesLoaded was still false; therefore the transition was not
run, [... cut]
commit d172a8e7 (2025-08-25): Brief: allow side panel to be closed when Brief is the Start Page

The side panel can only be closed when state="panelOpened", and this
state is set by the ScriptAction in the "initialized" -> "panelOpening"
transition. So this transition must be run whenever the state becomes
"panelOpening", regardless of whether Global.allPagesLoaded is true;
otherwise, it is possible to get stuck in the "panelOpening" state
without moving onto the "panelOpened" state.

This was happening when the Start Page was "Brief (side panel open)",
where Brief was loaded as the Start Page while Global.allPagesLoaded w [... cut]
commit b2fc5853 (2025-08-26): Brief: allow side panel to be closed when Brief is the Start Page

The side panel can only be closed when state="panelOpened", and this
state is set by the ScriptAction in the "initialized" -> "panelOpening"
transition. So this transition must be run whenever the state becomes
"panelOpening", regardless of whether Global.allPagesLoaded is true;
otherwise, it is possible to get stuck in the "panelOpening" state
without moving onto the "panelOpened" state.

This was happening when the Start Page was "Brief (side panel open)",
where Brief was loaded as the Start Page while Global.allPagesLoaded w [... cut]
closed 2025-08-26 by blammit