victronenergy/gui-v2 #2375: Settings -> Integrations page: Digital I/O entry sometimes missing

mpvader, 2025-07-23:
https://vrm.victronenergy.com/installation/795268/overview
v3.70~13

Seems to be a bug in gui-v2, since digital io is on the d-bus, and looking at this in gui-v1 works fine.

Next step is for user to try and reproduce with v3.70~19 once available later today, just in case the newer gui-v2 versions now available have already fixed this somehow.

com.victronenergy.digitalinputs on dbus:

[image]

page with missing entry, red line is where the entry is expected to show:

[image]

gui-v1 showing it:

[image]

one more dbus screenshot:

[image]

relevant source code:
https://github.com/victronenergy/gui-v2/blob/main/pages/settings/PageSettingsIntegrations.qml

blammit, 2025-07-24:
@mpvader that specific code hasn't changed since the v3.60 release - could you add access to the VRM installation please?

mpvader, 2025-07-24:
Done. And today the user will reproduce and then leave on on 24/7 .

System is now off

blammit, 2025-07-24:
In the gui-v2 code, the "Digital I/O" menu item appears when one or more `com.victronenergy.digitalinputs/Devices/<x>/Label` values are valid. I wonder if it's possible for those entries to sometimes be invalid even when the I/O devices are present.

(In gui-v1 it is done differently - the devices are read from `com.victronenergy.settings/Settings/DigitalInput` instead. It used be done like this in gui-v2, but it was updated as `com.victronenergy.digitalinputs` is now the correct place to find them.)

blammit, 2025-08-24:
The model checks for the existence of a valid `/Label` value as per `com.victronenergy.digitalinputs/Devices/<x>/Label`, but really it should work if there is any leaf node like `com.victronenergy.digitalinputs/Devices/<x>`. We can rework this once the model updates are implemented in #2371 .