victronenergy/gui-v2 #2380: Add VE.Bus Microgrid sub-menu

philipptrenz, 2025-07-24:
There is an upcoming feature for our Inverter/Chargers, which allows to have multiple systems jointly providing an AC micro-grid. This feature requires an additional sub-menu called "Microgrid parameters" under the VE.Bus product page.

When an external EMS is used to control the parameters, then gui-v2 is only used to monitor the parameters read-only. If not controlled externally, only the "Hybrid droop" mode is accessible, but then with all the different settings.

There are also charts planned that illustrate the current state of the system relative to it's setpoints – this is to be considered of lower priority and can follow later.

[Link to designs in Figma](https://www.figma.com/design/GjCtz3A3jqSGvyjVvOm6RZ/Venus-OS--GX-Touch-?node-id=11635-362959&t=pFfDETV5D17X2xAG-1)

philipptrenz, 2025-07-25:
@vejpasop has finished the documentation on the dbus paths (nicely done!) in above linked issue.

vejpasop, 2025-09-04:
Daniel has the VRM link to my test setup for testing.

commit 274894f8 (2025-09-04): Add VE.Bus Microgrid sub-menu

Fixes #2380
commit d88c8141 (2025-09-21): Add VE.Bus Microgrid sub-menu

Fixes #2380
commit cfdc44ba (2025-09-22): Add VE.Bus Microgrid sub-menu

Fixes #2380
commit 7a9dc40c (2025-09-22): Add VE.Bus Microgrid sub-menu

Fixes #2380