victronenergy/gui-v2 #2395: Electron flow direction is sometimes incorrect on Overview page

chriadam, 2025-07-30:
The geometry_overviewPage_connector_animationPowerThreshold theme constant is 30.
It is used in binding expressions of the form:

```
                        // For AC inputs, positive power means energy is flowing towards inverter/charger,
                        // and negative power means energy is flowing towards the input.
                        return power > Theme.geometry_overviewPage_connector_animationPowerThreshold
                                        ? VenusOS.WidgetConnector_AnimationMode_StartToEnd
                                        : VenusOS.WidgetConnector_AnimationMode_EndToStart
```

this ensures that the direction of the flow doesn't "flicker" when the power is close to zero.

However, the logic is incorrect, and it is particularly visible when animations are disabled (as the arrow is always shown, and in the error case, it points in the wrong direction).

These bindings should be improved to be something like:

```
return Math.abs(power) < Theme.geometry_overviewPage_connector_animationPowerThreshold ? NoDirection
    : power > 0.0 ? StartToEnd
    : EndToStart
```

mpvader, 2025-07-30:
When was this broken? Since forever? Or introduced recently?

mr-manuel, 2025-07-30:
Fixing this issue, fixes maybe also this https://github.com/victronenergy/gui-v2/issues/2360

chriadam, 2025-07-31:
> When was this broken? Since forever? Or introduced recently?

Since forever.  It became more "obvious" recently because of the "disable animations" thing, which always shows an arrow in the "bound" direction, even if the power level is lower than 30W.

commit a85a5565 (2025-07-31): Modify power flow indicator display

Modify how the animated energy flow is displayed.
When not animated, only show an arrow when power is above 30
in either direction.

Fixes #2395
PR #2399 (2025-08-12): Modify power flow indicator display
Modify how the animated energy flow is displayed.
When not animated, only show an arrow when power is above 30 in either direction.

Fixes #2395
commit 61d2105f (2025-08-13): Modify power flow indicator display

Modify how the animated energy flow is displayed.
When not animated disable the visibility.

Fixes #2395