victronenergy/gui-v2 #2411: Switchable output deadlock

martinbosma, 2025-08-07:
When an output is changed, but the new value is not allowed for some reason, the UI components gets in a deadlock.

Consider the following example with two toggle buttons:
- A main switch 
- A run switch that is only allowed to do something if the main switch is on

Initial state, main switch off and run switch off:
[image]

Then I try to set the run switch to on:
[image]

The `1` value has been written to `SwitchableOutput/n/State` but not accepted and remains `0`. Since the value didn't change, no value is sent back to the UI. However, now there is no way to change the run switch anymore. Do note that the WASM build behaves differently compared to the local UI. The local UI doesn't get in the "blocked" state, but no new values are sent either.

When I do force sent the unchanged value, it does behave a bit differently. It does reflect the actual state instead of the desired state but is still blocked.
[image]

It feels like perhaps it should at least always write the desired value in the button is pressed and/or there should be a timeout on the toggle state. Perhaps we also need to forcefully send the value from the dbus/mqtt service even if the value hasn't changed. This currently isn't the case. Also, this example applies to the toggle switch but likely affects other types as well.

This setup is available for testing: https://vrm.victronenergy.com/installation/496541/dashboard
However, the SwitchableOutput might not always be available. Please contact me when you need it.

blammit, 2025-08-14:
> I suppose preset the button state with the desired state is likely done because of fast feedback to the user. Otherwise, if you wait for a response from the device to set the active state, it might feel sluggish. 

Yes, exactly. The current settings list switches do not do this, and that does feel sluggish, especially in Wasm, so perhaps we'll need to consolidate the behaviours at some stage.

>  And would it help that, if the desired state is not accepted, to send the active value again (even if it didn't change)? Can both native and wasm builds handle that?

I'm not sure about this one. It is difficult to know when it's not accepted, at least in Wasm, because we need to wait for a value change, which means we will need some timeout, just like we do in this case here when we are trying to detect that here in terms of whether the state changes to the desired value. To reduce the complexity it might be safer to just send the value when the user explicitly re-clicks the button (when it is enabled again after the 5 seconds).

martinbosma, 2025-08-20:
Well, at least at the timeout. I've had a talk with one of the digital switching integration partners. They also have a web interface an use a timeout of around 250ms, so 3-5s seems way out of line compared to that. For local MQTT/dbus connection I suppose that would be ok, but when using remote console on VRM they can be quite some latency, I think.

What is you planning according to fixing this issue?

blammit, 2025-08-21:
We do have a timeout currently of 3 seconds for the VRM/remote case, though the concern there is more the latency rather than the value change not being accepted by the device. I'm doing some more testing on this today and will discuss in the dev meeting, and then should have a PR up soon after.

blammit, 2025-08-22:
> The other thing discussed was that Martin sometimes sees the control gray out, and sometimes not, which needs to be discussed with Rein. I'd prefer the control to never gray out, at least not under any normal circumstances.

Created https://github.com/victronenergy/gui-v2/issues/2439 for this.

blammit, 2025-10-02:
Changes merged; closing.

closed 2025-10-02 by blammit

martinbosma, 2025-10-03:
@blammit I have a python script that seems to simulate this behavior. I see with the latest version (v1.2.13) that the local UI behavior seems to have changed, yet it seems that once it is out of sync with the back-end, it no longer send new values if you press it.

`dummyswitch.py` goes into `/opt/victronenergy/dbus-systemcalc-py/scripts` on the GX.
```python
#!/usr/bin/env python3

from dbus.mainloop.glib import DBusGMainLoop
from gi.repository import GLib
import argparse
import logging
import sys
import os

# our own packages
sys.path.insert(1, os.path.join(os.path.dirname(__file__), '../ext/velib_python'))
from dbusdummyservice import DbusDummyService
from logger import setup_logging

# Argument parsing
parser = argparse.ArgumentParser(
    description='dummy dbus service'
)

parser.add_argument("-n", "--name", help="the D-Bus service you want me to claim",
                type=str, default="com.victronenergy.switch.socketcan_can0_vi0_uc12345")
parser.add_argument("-c", "--channels", help="the number of channels for the service",
                type=int, default=1)

args = parser.parse_args()

print(__file__ + " is starting up, use -h argument to see optional arguments")
logger = setup_logging(debug=True)

# Have a mainloop, so we can send/receive asynchronous calls to and from dbus
DBusGMainLoop(set_as_default=True)

paths = {}
for c in range(args.channels):
    paths.update({
        f'/SwitchableOutput/{c}/Settings/CustomName': {'initial': 'My switch'},
        f'/SwitchableOutput/{c}/Settings/DimmingMax': {'initial': 100},
        f'/SwitchableOutput/{c}/Settings/DimmingMin': {'initial': 0},
        f'/SwitchableOutput/{c}/Settings/FuseRating': {'initial': None},
        f'/SwitchableOutput/{c}/Settings/Group': {'initial': 'My group'},
        f'/SwitchableOutput/{c}/Settings/Labels': {'initial': ['option1', 'option2', 'option3'], 'writeable': False},
        f'/SwitchableOutput/{c}/Settings/ShowUIControl': {'initial': 1},
        f'/SwitchableOutput/{c}/Settings/StepSize': {'initial': 1},
        f'/SwitchableOutput/{c}/Settings/Type': {'initial': 1},
        f'/SwitchableOutput/{c}/Settings/Unit': {'initial': "V"},
        f'/SwitchableOutput/{c}/Settings/ValidTypes': {'initial': 2015},
        f'/SwitchableOutput/{c}/State': {'initial': 0, 'writeable': False},
        f'/SwitchableOutput/{c}/Status': {'initial': 0},
        f'/SwitchableOutput/{c}/Temperature': {'initial': None},
        f'/SwitchableOutput/{c}/Voltage': {'initial': None}
    })

s = DbusDummyService(
    servicename=args.name,
    deviceinstance=0,
    paths=paths,
    productname='Generic switching device',
    connection='VE.Can')

logger.info('Connected to dbus, and switching over to GLib.MainLoop() (= event based)')
mainloop = GLib.MainLoop()
mainloop.run()

```

You also need to patch `/opt/victronenergy/dbus-systemcalc-py/ext/velib_python/dbusdummyservice.py`
```diff
diff --git a/dbusdummyservice.py b/dbusdummyservice.py
index c070da7..055dea3 100755
--- a/dbusdummyservice.py
+++ b/dbusdummyservice.py
@@ -43,7 +43,9 @@ class DbusDummyService(object):
 
         for path, settings in self._paths.items():
             self._dbusservice.add_path(
-                path, settings['initial'], writeable=True, onchangecallback=self._handlechangedvalue)
+                path, settings['initial'], 
+                writeable=settings.get('writeable', True), 
+                onchangecallback=settings.get('onchangecallback', self._handlechangedvalue))
 
         self._dbusservice.register()
         GLib.timeout_add(1000, self._update)
```