victronenergy/gui-v2 #2479: Make the User access level less restrictive

mpvader, 2025-09-06:
How it should be is that when the access level is set to User, instead of User and installer, all operational controls remain accessible.

Only settings are to be blocked from changing.

Now when set to User only, the UI prevents turning the inverter on and off, and prevents changing the input current limit; and probably prevents access to a whole lot more things as well.

Note that the same rule applies to all other types of products, like chargers. On/off needs to be possible, and changing input current limit also.

Only setting, and in practice that only/mostly means GX device settings, are to be blocked from access.

gui-v1 is a good reference.

blammit, 2025-09-08:
> How it should be is that when the access level is set to User, instead of User and installer, all operational controls remain accessible.
> Only settings are to be blocked from changing.

By "operational controls outside of settings" it sounds like you're referring to those in:

- control cards 
- overview drilldowns

Some of the drilldowns go to the settings pages (e.g. product pages) so we'll need to check some of those cases.

The default write access level for settings is "installer" and this default is used for many settings, particularly those added in gui-v2. Perhaps of those new settings like the "Brief centre display" should also be user-writable too.

blammit, 2025-09-09:
Discussed this further. The access levels separate into support for "configuration" (for installers doing the initial setup) vs "operation" (for users, who operate it thereafter). Installers need to configure firmware, ess, dvcc, integration settings, etc. Users need to turn devices on/off, change current limit, display settings, etc.